duo-labs/EFIgy #2: http://efigy.io/ doesn't work

gcox, 2017-09-30:
The URL, http://efigy.io, you have in the repo's summary doesn't work.

swrobel, 2017-10-03:
Seems it just redirects to this github repo (via an https redirect first):

```
$ curl -I http://efigy.io
HTTP/1.1 301 Moved Permanently
Server: CloudFront
Date: Tue, 03 Oct 2017 18:15:52 GMT
Content-Type: text/html
Content-Length: 183
Connection: keep-alive
Location: https://efigy.io/
X-Cache: Redirect from cloudfront
Via: 1.1 86f2bbfea57d6217ce2d279b84ca3743.cloudfront.net (CloudFront)
X-Amz-Cf-Id: HkWHrmyoH-EbV4Oqd7o4b-Ts-_4hbDmGp86vp3hOaGXQK-MoBuHtWA==

$ curl -I https://efigy.io
HTTP/2 301 
content-length: 0
location: https://github.com/duo-labs/EFIgy/
date: Tue, 03 Oct 2017 18:16:17 GMT
server: AmazonS3
x-cache: Miss from cloudfront
via: 1.1 0826158c5b3316bf0a83b23783f3dc7a.cloudfront.net (CloudFront)
x-amz-cf-id: g8cvJBawAfpC1Y1Yp-FxqmMvb_-DzHrxQ5PLqlyT5hSEtjYhEa-kpw==
```

kylelady, 2017-10-04:
Confirmed.

closed 2017-10-04 by kylelady

gcox, 2017-10-04:
DNS propagation delay strikes again.