duo-labs/EFIgy #6: Newer than expected EFI firmware and build of 10.12.6.

DerekCurrie, 2017-10-04:
I ran the current build of EFIgy, failed the first two tests, but found that I had _newer_ than expected firmware and OS build. That's how I'm interpreting the results. I participated in Apple's AppleSeed beta-testing program for 10.12.6, which may be why.

Below are the full test results, followed by my interpretation notes.

> EFIgyLite API Information:
> 	API Version: 0.2
> 	Updated On: Oct 3 2017, 16:44
> 
> 
> Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version): 
> 	Hashed SysUUID   : 6c300eb55df872a187804496eea3e07762c6a4c0bafb3b4b5a4dff0ad0a014ec
> 	Hardware Version : MacBookPro11,3
> 	Boot ROM Version : MBP112.0142.B00
> 	SMC Version      : 2.19f12
> 	Board-ID         : Mac-2BD1B31983FE1663
> 	OS Version       : 10.12.6
> 	Build Number     : 16G1023
> 
> [?] Do you want to continue and submit this request? [Y/N]  y
> 
> EFI firmware version check:
> 	[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro11,3) and OS build you have installed (16G1023). Your firmware MBP112.0142.B00, expected firmware MBP112.0138.B25.
> Update your firmware!!
> 
> Highest build number check:
> 	[-] ATTENTION - You are NOT running the latest build number of your OS version (10.12.6). Your build number is 16G1023, the latest build number is 16G29
> 
> Up-to-date OS check:
> 	[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.12.6)

1) My results show that my MacBookPro11,3 FAILed the EFI firmware test because it has NEWER firmware than was expected. Meanwhile, EFIgy tells me to ‘update’ my firmware. Wrong.

2) The “Highest build number check” of 10.12.6 FAILed. This one is very odd. I’d been AppleSeed testing 10.12.6. Therefore, it would not be strange for me to have a different build of it on my MBP. However, I’d installed the FULL public installation of 10.12.6 anyway. That means I’m supposed to have the full, final, finished version of 10.12.6 on the MPB. So why do the test results say:

> [-] ATTENTION - You are NOT running the latest build number of your OS version (10.12.6). Your build number is 16G1023, the latest build number is 16G29

And how do I interpret this FAIL? If I have build 16G1023, isn’t that well beyond the ‘latest’ 16G29? I think so!

zigg, 2017-10-04:
@DerekCurrie I have exactly the same results. Since you're on a beta channel, did you perchance install the Security Update Public Beta 2017-001?

DerekCurrie, 2017-10-04:
Below are results after re-installing 10.12.6 via the Repair volume on my MacBookPro11,3. Note that this is a different installer than I had previous used, which was the Combo update for 10.12.6. I'll discuss the possible repercussions below. As a result, the test for 10.12.6 build versions passed.

> EFI firmware version check:
> 	[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro11,3) and OS build you have installed (16G29). Your firmware MBP112.0142.B00, expected firmware MBP112.0138.B40.
> Update your firmware!!

Note that I still have the same firmware version. Note that it is still a higher version than what was expected. But also note that the test's 'expected' firmware versions have changed! Why? It points to the test being faulty. Originally expected was ...B25. Now expected ...B40. Huh?!

**What I think is going on:**

• Duo Labs didn’t take into account the build number difference for the **Combo** update of 10.12.6. That would explain the higher build number I had in the first test.

• Duo Labs does NOT have a complete list of EFI Firmware versions. I clearly have a MORE RECENT version of the firmware than they expect.

• Duo Labs still has work to do on its flaky software. Having a mismatch between test results regarding expected firmware versions is itself a FAIL.

Conclusion: This version of EFIgy and its corresponding web database are a source of confusion and not much else. Highly disappointing. I _cannot_ recommend anyone bother with it at this point. Sorry.

DerekCurrie, 2017-10-04:
@zigg: 

> did you perchance install the Security Update Public Beta 2017-001?

That's unlikely. I am only aware of having installed successive beta versions of specifically 10.12.x. I stayed entirely out of 10.13.x beta testing.

zigg, 2017-10-04:
@DerekCurrie It was a security update to 10.12.6. Looks like it was followed by a released patch to 10.12.6? https://support.apple.com/en-us/HT207835

[image]

DerekCurrie, 2017-10-04:
> @DerekCurrie <https://github.com/derekcurrie> It was a security update to 10.12.6. Looks like it was followed by a released patch to 10.12.6? https://support.apple.com/en-us/HT207835 [...]

I don’t recall. If it was a public security patch AFTER 10.12.6 was publicly released, I would have installed it. But that’s not what you’re asking, from what I can tell. I don’t know anything about a ‘beta’ 10.12.6 security patch that would have come out after 10.12.6 was publicly released.

Note that at this time I have reinstalled 10.12.6 twice after the initial 10.12.6 update. The first time was via the Combo installer. The second time, today, was via what’s fed to the Repair volume on my MacBookPro11,3. The firmware version remained the same between these reinstallations. I would assume that because I have a higher number, the installers would not attempt to install the lower numbers.

Also note that the EFIgy tests didn’t provide consistent ‘expected’ versions of firmware. I elaborated in my added notes at GitHub. That’s bad, resulting in confusion.

:-Derek Currie

————————————
Derek G Currie
derekcurrie@mac.com
derekcurrie@lavabit.com
————————————

bmeike, 2017-10-04:
I believe I have a similar issue.  The output of the checker, in part, is:
```
	[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro11,3) and OS build you have installed (15G1611). Your firmware MBP112.0138.B40, expected firmware MBP112.0138.B25.
Update your firmware!!

```
It seems likely to me that (though I absolutely do not no for sure) that .B40 is more recent that .B25.

alvarnell, 2017-10-04:
> derekcurrie@mac.com <mailto:derekcurrie@mac.com>
> derekcurrie@lavabit.com <mailto:derekcurrie@lavabit.com>

Yes, Security Update 2017-001 for Sierra came out on or about Sep 27th for testers. I expect it to be released along with macOS 10.13.1.

-Al-

-Al-
-- 
Al Varnell
Mountain View, CA

DerekCurrie, 2017-10-04:
Ah! I figured out the situation. On my end the update is entitled “Security Update Developer Beta 2017-001” with “Developer” being the variant word. I assume this update is exactly what you’re asking about.

IOW: YES, I did!

Therefore, let’s consider that the likely source of the newer EFI firmware.

Sorry for the confusion at my end.

:-Derek

> On Oct 4, 2017, at 2:44 PM, Matt Behrens <notifications@github.com> wrote:
>
> @DerekCurrie <https://github.com/derekcurrie> It was a security update to 10.12.6. Looks like it was followed by a [...]

————————————
Derek G Currie
derekcurrie@mac.com
derekcurrie@lavabit.com
————————————

bmeike, 2017-10-05:
I am less clear on why I have a more recent EFI than is expected.  I’m running 10:11:6.  I see the following two updates, that may be relevant:

                <string>Security Update 2017-003</string>
                <string>10.11.6</string>

… and:
                <string>macOS</string>
                <string>12.6.03</string>

Hope that helps…

 - blake

kkumlien, 2017-10-05:
I've a similar issue. Why is it asking me to update if I have a newer version?

Thanks!

```
EFI firmware version check:
	[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro11,3) and OS build you have installed (14F2511). Your firmware MBP112.0138.B18, expected firmware MBP112.0138.B16.
Update your firmware!!
```

SonicPurity, 2017-10-05:
Similar. EFIgy is happy with the Sierra version, but not my 2012 15" non-Retina MacBook Pro’s firmware:

EFI firmware version check:
	[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro9,1) and OS build you have installed (16G29). Your firmware MBP91.00D7.B00, expected firmware MBP91.00D3.B22.
Update your firmware!!

Now unless i’m counting wrong, 00D7.B00 is higher than 00D3.B22.

Maybe it’s a matter of wording. The first part says “unexpected” which may well be true. It’s the last sentence “Update your firmware!!” which appears false.

Note to developers: i ran EFIgy with --debug and saved the entire results. Let me know if you want it, and how to get it to you.

Note 2: I usually only use general release software and trail a year behind Apple (e.g. i just recently upgraded to Sierra). However this year i did participate in the High Sierra Public Beta with this Mac, lastly with the GM candidate. That was on an external test drive, then i reverted to stock Sierra for regular production work on this Mac.

alvarnell, 2017-10-05:
@SonicPurity, As indicated, the tool has not been updated for macOS High Sierra yet, so it knows nothing about the fact that you updated your firmware by running HS. It's also clear that the tool isn't for use by anybody that has participated in a public (or otherwise) beta test since those updates haven't been included and my guess is they won't in the future. It's mostly for Enterprise administrators who need to check machines that are on a regular update track.

MyNameIsMeerkat, 2017-11-22:
As of commit https://github.com/duo-labs/EFIgy/commit/798116c0b5bd827b04febbfd156f4b44ff673985 there is now initial 10.13.x support in the dataset and firmware that appears *newer* than expected is better messaged to the user.

closed 2017-11-22 by MyNameIsMeerkat

kkumlien, 2018-02-05:
https://check.efigy.io/ still says that "macOS 10.13 is NOT currently supported". Is that true?

alvarnell, 2018-02-06:
Note that ‘this issue is marked closed, so you may not get a response and need to open a new issue wupith details on your platform, etc.