duo-labs/EFIgy #7: EFI not identified on Mac7,1?

HarryFJ, 2017-10-04:
I don't understand what this error-message means: "Can't get X86PlatformPlugin"

Log:

Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version): 
2017-10-04 22:05:57.458 system_profiler[18506:1078164] platformPluginDictionary: Can't get X86PlatformPlugin, return value 0
2017-10-04 22:05:57.460 system_profiler[18506:1078164] platformPluginDictionary: Can't get X86PlatformPlugin, return value 0
	Hashed SysUUID   : 8156bb3a4413f6652078137070d340070be58c8669a09f9633f1a8a3f01abb64
	Hardware Version : iMac7,1
	Boot ROM Version : IM71.007A.B03
	SMC Version      : 1.20f4
	Board-ID         : Mac-XXXXXXXX
	OS Version       : 10.10.5
	Build Number     : 14F2511

[?] Do you want to continue and submit this request? [Y/N]  Y

EFI firmware version check:
	[-] ATTENTION - Your Mac model (iMac7,1) does not seem to have had any EFI updates released for it :'(

Highest build number check:
	[+] SUCCESS - You are running the latest build number (14F2511) of the OS version you have installed (10.10.5)

Up-to-date OS check:
	[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.10.5)

logout

Thanks, Harry

MyNameIsMeerkat, 2017-10-05:
Hi Harry,

So the `X86PlatformPlugin` error seems to an issue with the native OS X `system_profiler` system command that the EFIgyLite_cli script uses to gather some of the version data it sends to the API. From other reports of it I can find it looks like something that is specific to the `system_profiler` that ships with 10.10 and a quick look around didn't give a clear consensus as to  what causes it. So while the error is not a bug with EFIgy itself, the spurious output that is confusing the rest of the output from the client is definitely something that should be taken care of. 

I can't reproduce the error myself so I took a bit of a shot in the dark that suppressing the stderr output of the `os.Popen()` that is used to call to the `system_profiler` command itself may stop the confused output you're seeing (but won't address the root cause of the error of course).

As I am unable to reproduce the issue if you could test with the latest version that I have committed (diff [here](https://github.com/duo-labs/EFIgy/commit/334c096307c6a003cd67a5c00915d163a1d79123#diff-eebe91fe7320688945745dcc1feb0fb6) and let me know whether you're still seeing the errors in the `EFIgyLite_cli` output that would be awesome.

HarryFJ, 2017-10-05:
Hello MyNameIsMeerkat,

yes, output is suppressed, but yes, root problem not solved and EFI-version not recognized. This is the new output:

Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version): 
	Hashed SysUUID   : 8156bb3a4413f6652078137070d340070be58c8669a09f9633f1a8a3f01abb64
	Hardware Version : iMac7,1
	Boot ROM Version : IM71.007A.B03
	SMC Version      : 1.20f4
	Board-ID         : Mac-XXXXXXXX
	OS Version       : 10.10.5
	Build Number     : 14F2511

[?] Do you want to continue and submit this request? [Y/N]  Y

EFI firmware version check:
	[-] ATTENTION - Your Mac model (iMac7,1) does not seem to have had any EFI updates released for it :'(

Highest build number check:
	[+] SUCCESS - You are running the latest build number (14F2511) of the OS version you have installed (10.10.5)

Up-to-date OS check:
	[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.10.5)

logout

MyNameIsMeerkat, 2017-10-08:
Hi HarryFJ,

The output you show is EFIgy working as expected, our data shows that the iMac7,1 model has never received an EFI firmware update from Apple and the output messages that. There is no problem here other than Apple not having released EFI firmware updates for your Mac hardware.

HarryFJ, 2017-10-09:
OK, thank you for your info!

Bye Harry

closed 2017-10-09 by HarryFJ