duo-labs/EFIgy #8: Don't rely on Korey for board-id

Piker-Alpha, 2017-10-05:
This is how you do it: https://github.com/Piker-Alpha/HandyScripts/blob/master/efiver.py#L329

Piker-Alpha, 2017-10-05:
You can do the same for src-version with

```
def getMySMCVersion():
	return IORegistryEntryCreateCFProperty(IOServiceGetMatchingService(0, IOServiceMatching("AppleSMC")), "smc-version", None, 0)
```

MyNameIsMeerkat, 2017-10-06:
Thanks, and 100% agree. 
Have updated the file to use IOKit/objc to get all of the base versions for EFI, SMC, and the Board_ID. Also updated the collection of sysUUID and the mac model to use objc calls as well with the following incase they're of use to you:

```
mac_model_code = IORegistryEntryCreateCFProperty(IOServiceGetMatchingService(0,IOServiceMatching("IOPlatformExpertDevice")), "model", None, 0)

sys_uuid = IORegistryEntryCreateCFProperty(IOServiceGetMatchingService(0, IOServiceMatching("IOPlatformExpertDevice")), "IOPlatformUUID", None, 0)
```

Appreciate the notes and helping to make the client better :)

closed 2017-10-06 by MyNameIsMeerkat