duo-labs/EFIgy #10: Fatal error unknown error (_ssl.c:2825)

a2X4, 2017-10-10:
Sorry, my Python skills are weak. Here is the --debug.
On an iMac running Version 10.12.6 (16G29)

 python EFIgyLite_cli.py --debug
DEBUG:__main__:[-] Certifi module not found, falling back to bundled cecert.pem file
DEBUG:__main__:[+] cacert file location: '/private/tmp/cacert.pem'

EFIgyLite API Information:
[-] Fatal error in EFIgyLite_cli. Exiting.....

Error:
	unknown error (_ssl.c:2825)

Traceback (most recent call last):
  File "EFIgyLite_cli.py", line 454, in <module>
    efigy_cli()
  File "EFIgyLite_cli.py", line 244, in __call__
    api_version = self.__make_api_get("/version")
  File "EFIgyLite_cli.py", line 156, in __make_api_get
    self.last_response = urllib2.urlopen(self.api_server+api_path, cafile=self.cacert_path)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib2.py", line 144, in urlopen
    capath=capath)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 435, in create_default_context
    context.load_verify_locations(cafile, capath, cadata)
SSLError: unknown error (_ssl.c:2825)

mdmilgrom, 2017-10-11:
I believe I had this issue and solved it by installing openssl via Homebrew. (brew install openssl)
If you don't have Homebrew installed, do that first.

MyNameIsMeerkat, 2017-11-22:
@a2X4 It's definitely an underlying SSL lib issue but would need more info to get it better diagnosed for you. Check the format of the .pem file that the EFIgy client is trying to use, and maybe try to just install the certifi module so as you can use the PEM's it points at.

If those don't fix the issue please provide some more info on your python and system versions etc

MyNameIsMeerkat, 2018-03-01:
No activity on this so closing out

closed 2018-03-01 by MyNameIsMeerkat