duo-labs/EFIgy #9: Security question

nark, 2017-10-10:
Hello there, I don't know if it is really the place to talk about this but… As serious and secure as Duo Labs could be, how can we trust the dataset information and results provided through the API ? 

I mean, a malicious user could easily fake some data from the inside, or your service integrity could be corrupted by attacks from the outside, and then propagating false positive results with a very large scale implications. I know for instance many IT companies and administrations that will refuse to use a such tool because it relies on external services, only seeing another attack vector to manage. 

So, what ensure to us that your service and data it provides are safe ? How well are protected your facilities and procedures ? What measures are being taken to avoid code and/or data corruption internally ? Do you plan to release a sort of offline dataset ?

I'm very sorry to make myself the devil's advocate, but from my point of view, these questions are very serious. The EFI version issue is real, but to rely on a service that is potentially another source of security breakdown is not a good enough solution.

MyNameIsMeerkat, 2017-11-22:
Hola @nark 

So lots of questions in here so I'll try and seperate them out a little:

* How can you trust Duo?
Quite honestly trust is yours to determine where you want to place it. I'm sure you use code that is from a variety of open source projects and online sources each day, follow the same rules you do for those sources to determine whether you trust Duo and the data we have compiled and provide for free.

* How can you trust the data in the API?
Again, trust is yours to place where you see appropriate. The data is compiled on a best effort basis and as far as we know it is the only source of Apple EFI version data available, but there is always the chance for there to be a mistake in the dataset and we know for sure it is by definition incomplete. If the API, dataset and tools are helpful to you then great, if you don't trust them enough to use them then that's your decision to take and we completely understand.

* How can we be sure the services we provide are safe?
The client is open source so feel free to inspect the code and let us know questions or bugs you find. The API is a very simple RESTful API making on the wire inspection of requests and responses easy to review. From your review of the client code you will see that the responses from the API are just JSON messages that get parsed and provide a guidance message to the user nothing more, as such if the API server got _hacked_ it could give you incorrect data in response but couldn't cause the client to 'do anything bad'. Again exercise the same level of scrutiny and caution with this service as you would for any other API you make use of in your day to day computer use.

* Protections against internal corruption/the NSA zip'lining into the office at night to mess with the dataz.....
This is an opensource project and is provided on a best effort basis. There are no guarantees as to the correctness of the dataset, but we do our best to have it as accurate as possible. Our threat model does not currently include protections against the NSA, CIA or even MI6. 

* Do we plan on releasing an offline version of the dataset?
I won't say never but we don't have a timeline on doing so at the moment. The online API allows us to keep the dataset up to date for everyone as easily as possible and also allows to continue to gather data about the versions of EFI running on Apple systems that contributes towards our continued research in the space. If your organisation's threat model doesn't allow you using an online API then I would suggest building up your own EFI version dataset using the detailed walkthrough we gave in the technical paper and using that offline so as you can keep a check on the EFI versions being run.

MyNameIsMeerkat, 2018-03-01:
No activity on this for a while so closing it out

closed 2018-03-01 by MyNameIsMeerkat