duo-labs/EFIgy #11: Unknown build number

chrish42, 2017-10-11:
When running EFIgy on my system, I get the following error:

EFI firmware version check:
	[!] ERROR - Unknown Build Number '15G1611' given. (1507685319.35)

JoesTechRoom, 2017-10-11:
I also got this message:

EFIgyLite API Information:

API Version: 0.2

Updated On: Oct 3 2017, 16:44

Enumerated system informaton (This data will be sent to the API in order to
determine your correct EFI version):

Hashed SysUUID   :
8c90892b6981b99f6f96cf84fca7b4d02ad74354e403a1d9534e8b7d2ca08f01

Hardware Version : Macmini4,1

EFI Version      : MM41.0042.B03

SMC Version      : 1.65f2

Board-ID         : Mac-F2208EC8

OS Version       : 10.11.6

Build Number     : 15G1611

[?] Do you want to continue and submit this request? [Y/N]  Y

EFI firmware version check:

[!] ERROR - Unknown Build Number '15G1611' given. (1507580535.37)

Highest build number check:

[+] SUCCESS - You are running the latest build number (15G1611) of the OS
version you have installed (10.11.6)

Up-to-date OS check:

[+] SUCCESS - You are running the latest major/minor/micro version of the
OS you have installed (10.11.6)

MyNameIsMeerkat, 2017-11-22:
Interesting.

@JoesTechRoom With the output you shared I can see the hardware model and OS version you are running, and that combination is not one that we actually have much data for. There was no MM41 EFI firmware shipped in OS build 15G1611 (10.11.6 SecUpdate 2017-003) and there was no EFI firmware released in any other previous 10.11.x update - so that explains the error message as the server doesn't know any valid version of EFI to recommend as we basically have zero datapoints for this hardware/os build combo.
My guess is that the version of EFI you are running (MM41.0042.B03) is what the system left the factory with as we have seen no in-field updates for that hardware from Apple. 

@chrish42 I don't have as much info to go on for you but I expect the same is probably true. 

Long term I would love to get to the place where we have the data of the EFI versions different hardware left the factory with, but for right now that is not data that we have a complete enough picture of to include. I'm going to think about if there's a nice way for us to get this data .....

Thanks for raising these questions

closed 2017-11-22 by MyNameIsMeerkat