duo-labs/EFIgy #12: log option

bc75, 2017-10-11:
The -l option doesn't work as indicated in the output of -h.
-h shows 
" -l LOG, --log LOG  File to log output to"
-l is treated as directory that the output file is written to with a generated filename.

Running the following command
" python EFIgyLite_cli.py -l efigy-out-2017-10-11.txt "
resulted in the following error with "N" replacing the the actual value generated
"[!] Error opening specified log file at 'efigy-out-2017-10-11.txt/EFIgyLite_cli_NNNNNNNNNNN_NN.log' - [Errno 2] No such file or directory: 'efigy-out-2017-10-11.txt/EFIgyLite_cli_NNNNNNNNNNN_NN.log'
"

jholt13, 2017-10-19:
When I designate `.` as the destination following `-l` a log is created in the local directory.  Perhaps the command line args could be updated to specify that the tool will output to a directory instead of a file.

MyNameIsMeerkat, 2017-11-22:
Yup, I've cleaned up the -h messaging to be clearer that -l sets the directory to log to and is not the name of the file to log to. 

Thanks raising

closed 2017-11-22 by MyNameIsMeerkat