duo-labs/EFIgy #13: EFI firmware version check ERROR - Unknown Build Number '17B48' given. (1511394575.64)

justPaulo, 2017-11-22:
[pnascim@MacMini-do-Paulo ~]$ python /Users/pnascim/Downloads/EFIgy-master/EFIgyLite_cli.py 

EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Nov 22 2017, 14:05

------------------------------------------------------------------------------------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : 6e6c1d07b2705fa2f9be7d1bd69528c6dd310f0d33b7840a1e7f4119446d2937
	Hardware Version : Macmini6,2
	EFI Version      : MM61.010B.B00
	SMC Version      : 2.8f0
	Board-ID         : Mac-F65AE981FFA204ED
	OS Version       : 10.13.1
	Build Number     : 17B48

[?] Do you want to continue and submit this request? [Y/N]  Y

	# Results:

	EFI firmware version check:
		[!] ERROR - Unknown Build Number '17B48' given. (1511394575.64)

	Highest build number check:
		[+] SUCCESS - You are running the latest build number (17B48) of the OS version you have installed (10.13.1)

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.13.1)

MyNameIsMeerkat, 2017-11-23:
Yeah not good! This was an error on the server where an incomplete dataset was being used for the mappings of hardware which shares EFI firmware (which is why your Macmini6,2 triggers the issue).

I've fixed the dataset and pushed the version to the server so it should be fixed now. Let me know if you continue to have a problem.

Thanks for raising the issue

justPaulo, 2017-11-23:
Hate to contradict, but still don't see improvements:

[theUser@MacMini-do-Paulo EFIgy-master]$ ./EFIgyLite_cli.py  --debug
DEBUG:__main__:[-] Certifi module not found, falling back to bundled cecert.pem file
DEBUG:__main__:[+] cacert file location: '/Users/theUser/Downloads/EFIgy-master/cacert.pem'

EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Nov 22 2017, 14:05

-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : 6e6c1d07b2705fa2f9be7d1bd69528c6dd310f0d33b7840a1e7f4119446d2937
	Hardware Version : Macmini6,2
	EFI Version      : MM61.010B.B00
	SMC Version      : 2.8f0
	Board-ID         : Mac-F65AE981FFA204ED
	OS Version       : 10.13.1
	Build Number     : 17B48

[?] Do you want to continue and submit this request? [Y/N]  Y

	# Results:

	EFI firmware version check:
[-] Fatal error in EFIgyLite_cli. Exiting.....

Error:
	invalid literal for int() with base 10: '010B'

Traceback (most recent call last):
  File "./EFIgyLite_cli.py", line 625, in <module>
    efigy_cli()
  File "./EFIgyLite_cli.py", line 367, in __call__
    self.check_fw_versions(sys_info, api_results)
  File "./EFIgyLite_cli.py", line 539, in check_fw_versions
    api_efi_ver = int(api_efi_str[1])
ValueError: invalid literal for int() with base 10: '010B'

[theUser@MacMini-do-Paulo EFIgy-master]$

MyNameIsMeerkat, 2017-11-23:
So this is why you shouldn't push fixes the day before Thanksgiving when the beverages have already started to flow .....

The issue was that I hadn't accounted for certain EFI versions are denoted in hex form, I learnt to count to 16 and fixed the issue in this commit https://github.com/duo-labs/EFIgy/commit/316d70bf5e7a4a40074f13e5a414162781145eb8. 

I tested by mimicking your system's versions and everything looked good here so it should work for you now.

Thanks for raising the issues you find, much appreciated! 🦃🦃🦃

justPaulo, 2017-11-23:
LoL, everybody knows that no gitting or coding should be done in (Thanksgiving)-1, page 723... :-)
You're welcome, thanks for the turkeys.

[balthazar@MacMini-do-Paulo EFIgy-master]$ date; ./EFIgyLite_cli.py 
Thu Nov 23 23:10:34 WET 2017

EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Nov 22 2017, 14:05

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : 6e6c1d07b2705fa2f9be7d1bd69528c6dd310f0d33b7840a1e7f4119446d2937
	Hardware Version : Macmini6,2
	EFI Version      : MM61.010B.B00
	SMC Version      : 2.8f0
	Board-ID         : Mac-F65AE981FFA204ED
	OS Version       : 10.13.1
	Build Number     : 17B48

[?] Do you want to continue and submit this request? [Y/N]  Y

	# Results:

	EFI firmware version check:
		[+] SUCCESS - The EFI Firmware you are running (MM61.010B.B00) is the expected version for the OS build you have installed (17B48) on your Macmini6,2

	Highest build number check:
		[+] SUCCESS - You are running the latest build number (17B48) of the OS version you have installed (10.13.1)

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.13.1)

closed 2017-11-23 by justPaulo