duo-labs/EFIgy #14: API Feature request

bochoven, 2017-11-24:
I like the `/apple/latest_efi_firmware/` endpoint better than the `/apple/oneshot` endpoint (due to the complex JSON you need to POST)

It would be nice if we could send a GET request to the API with:

- hw_ver
- build number
- Boot ROM Version (EFI)

and it would return JSON with something like: 'older', 'uptodate', 'newer' or 'notfound', so that we don't have to do the comparison of the EFI version on the client.

MyNameIsMeerkat, 2018-03-01:
Seems like a good suggestion I've added this to the feature request list and we will get to it when we can.

This issue will be updated with progress of the feature as it's made

MyNameIsMeerkat, 2018-03-09:
The new API endpoint and demonstration of it's use in the client have now been released in this commit https://github.com/duo-labs/EFIgy/commit/2ca33213e6e11a5ef4e4543341a974a6f98606c6 

Let us know if this does what you needed

closed 2018-03-10 by MyNameIsMeerkat