duo-labs/EFIgy #15: EFI version not found

bochoven, 2017-11-24:
macOS: 10.10 (14A389)
EFI: MBP71.0039.B0E
HW: MacBookPro7,1

MyNameIsMeerkat, 2017-11-25:
Hi there,

We don't have any data for the MacBookPro7,1 on 10.10.x or 10.11.x as our analysis shows there have been no field updates to the EFI of that model in any OS or Security updates. This means this is likely the firmware that your macbook left the factory with. 

Your build number of 14A389 show's you are running the base build of 10.10.0 which is now very out of date and vulnerable to a lot of security issues, on this system old EFI firmware is probably the least of your security concerns to be honest.

Having said this I've taken a feature note to add in better user messaging for models of hardware we know haven't seen in-field EFI updates, I'll push a new version of the client when I get a chance that makes it more obvious when this is the case.

bochoven, 2017-11-25:
This was just an example, I have a database with 4000+ Macs that Iwhich I partly shared with Pepijn. I know the machine is not very up to date. If you want, I can share the data that I have.

groob, 2017-11-28:
getting 
{"latest_efi_version": {"msg": "Unknown model of Apple Mac MacBookPro11,5  given. Provide models in the form of MacPro6,1 for Mac Pro (Late 2013)", "timestamp": "1511865528.96", "error": true}, "latest_os_version": {"msg": "10.13.1"}, "latest_build_number": {"msg": "17B48"}}

bruienne, 2018-03-09:
Just dropping in here to hopefully be able to close this issue:  @bochoven - The MBP7,1 is a legacy model and has not seen EFI updates after 2012 and thus never got any patches for Thunderstrike and beyond. We report back with warnings for legacy models to indicate this.

Separately @groob - updates have been rolled out since you tacked on your issue here, I assume it's been well resolved by now but LMK if it's still an issue - I will close this one and we can open a new issue to deal with any remaining incorrect model errors.

closed 2018-03-09 by bruienne

bochoven, 2018-03-09:
Thanks for the follow up!