duo-labs/EFIgy #16: Dubious Version Number Warning

herzi, 2017-12-06:
When I run the command, I get (among other information) this line:

     EFI firmware version check:
		[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro11,5) and OS build you have installed (16G1036). Your firmware is MBP114.0177.B00, the firmware we expected to see is MBPro114.0177.B00.

It looks dubious to me as the only difference in the version numbers is “MBP” vs. “MBPro” (where the “P” should represent the pro already).

mdmilgrom, 2017-12-06:
Similar results for MacBookAir:
Endpoint: 127.0.0.1
	# Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : ae019064a30a56eda29b9e2bf5ae65159e8b999c1d1c4b6b539fe042245a086e
	Hardware Version : MacBookAir5,2
	EFI Version      : MBA51.00F4.B00
	SMC Version      : 2.5f9
	Board-ID         : Mac-2E6FAB96566FE58C
	OS Version       : 10.13.1
	Build Number     : 17B1003

[?] Do you want to continue and submit this request? [Y/N]  y

	# Results:

	EFI firmware version check:
		[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookAir5,2) and OS build you have installed (17B1003). Your firmware is MBA51.00F4.B00, the firmware we expected to see is MBAir51.00F4.B00.

	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.13.1). Your build number is 17B1003, the latest release build number is 17B48

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.13.1)

mdmilgrom, 2017-12-06:
Also, this only works with the Mac native python install (/usr/bin/python), not with Anaconda python3. This should probably be documented somewhere, or the code changed to include a shebang for /usr/bin/python?

lindhor, 2017-12-07:
Also getting similar results on my MacBook Air mid-2013. 
```
EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Nov 22 2017, 14:05

--------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system informaton (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : e9113313cb53847239414294e5316256d02b4d20fe1c13863401072bbcf74cf7
	Hardware Version : MacBookAir6,2
	EFI Version      : MBA61.0103.B00
	SMC Version      : 2.13f15
	Board-ID         : Mac-7DF21CB3ED6977E5
	OS Version       : 10.13.1
	Build Number     : 17B1003

[?] Do you want to continue and submit this request? [Y/N]  y

	# Results:

	EFI firmware version check:
		[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookAir6,2) and OS build you have installed (17B1003). Your firmware is MBA61.0103.B00, the firmware we expected to see is MBAir61.0103.B00.

	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.13.1). Your build number is 17B1003, the latest release build number is 17B48

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.13.1)
```

bruienne, 2018-03-09:
@herzi @mdmilgrom @lindhor - We found the issue that could cause the issue you're seeing in certain corner cases and patched the API to fix it. If you could, please test whether the returned results for your endpoints are now as expected. I'll go ahead and close this once I hear back with a 👍 or 👎 Thanks for your reports!

@mdmilgrom You are correct, the tool is meant to work with stock system Python. We will update the shebang to use the system `/usr/bin/python` stub instead. Thank you for reporting.

The dataset is up to date with macOS 10.13.3 Supplemental Update (build 17D102 for non-iMac Pro and 17D2102/17D2104 for iMac Pro endpoints).

mdmilgrom, 2018-03-10:
I haven't seen it recur.
But I noticed "informaton" is misspelled in the above message.

MyNameIsMeerkat, 2018-03-10:
Glad it's not still occurring for you @mdmilgrom and thanks for spotting the typo, it's fixed in https://github.com/duo-labs/EFIgy/commit/3dc9315469f4594c46d9cff488d73251844beb38 

Shebang was updated in this commit https://github.com/duo-labs/EFIgy/commit/e75c95d07f9295943cb1a9ab281bd3897d9d9cc5 

Closing this one out as the problems seem to have been addressed, if the issues persist or reappear feel free to resurrect this

closed 2018-03-10 by MyNameIsMeerkat