duo-labs/EFIgy #17: Incorrect results?

Lloir, 2017-12-07:
# Results:

	EFI firmware version check:
		[-] ATTENTION - You are running an unexpected firmware version given the model of your system (MacBookPro9,2) and OS build you have installed (16G1114). Your firmware is MBP91.00D7.B00, the firmware we expected to see is MBPro91.00D7.B00.

	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.12.6). Your build number is 16G1114, the latest release build number is 16G1036

iMatthewCM, 2017-12-13:
I'm seeing this too. It looks like the difference is EFIgy is expecting "MBPro91.00D7.B00" and the system is returning "MBP" and then the exact same numbers. So, the firmware looks like it IS correct, but EFIgy is expecting the wrong thing.

G60GT, 2018-01-14:
Same issue with the build number on fully patched 10.13.2 system with supplemental update applied:

> 	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.13.2). Your build number is 17C205, the latest release build number is 17C88

bruienne, 2018-03-08:
Hey there! We found the issue that could cause out of date results in certain corner cases and patched the API to fix it. If you could, please test whether the returned results for your endpoints are now as expected.

The dataset is up to date with macOS 10.13.3 Supplemental Update (build 17D102 for non-iMac Pro and 17D2102/17D2104 for iMac Pro endpoints). Thanks for your reports!

bruienne, 2018-03-10:
@Lloir @G60GT @iMatthewCM When you get a chance let us know whether the issue still exists after aforementioned API update so we can go ahead and close this.

G60GT, 2018-03-10:
I do get the expected results now on my fully patched system. Thank you!

MyNameIsMeerkat, 2018-03-11:
Excellent to hear, we'll close out this issue. Thanks for letting us know about the problem

closed 2018-03-11 by MyNameIsMeerkat