duo-labs/EFIgy #19: Incorrect Results

Mraccountname, 2018-01-04:
EFI firmware version check:
		[-] ATTENTION - Your Mac model (iMac7,1) does not seem to have had any EFI updates released for it :'(

	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.11.6). Your build number is 15G18013, the latest release build number is 15G17023

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.11.6)

First of all, my Boot ROM version is IM71.007A.B03, which is the latest for the iMac 7,1 (I looked it up and tried to update it). Also, 15G18013 is the latest build, not 15G17023.

bruienne, 2018-03-09:
Hi there! Regarding the lack of available EFI updates for your Mac model (iMac7,1) this is because Apple never shipped a newer version after the one from 2008 listed on the page you referenced. Our data starts with updates released in 2015, after Apple changed the deployment method in response to the Thunderstrike vulnerability and beyond. Because of this we had to consider yours and other models legacy as there is no data to track for them.

To add some more context, any response you see for this model will contain a warning which is an expected behavior: to alert on old unsupported Mac models that can not be trusted if your organization's threat model includes nation-state firmware attacks. It is up to the reader to determine whether this is the case for them or not.

As for the incorrect build being reported, we have made some tweaks to resolve this. The OS build data is up to date with 15G19009 as of January 23, 2018. If you could, please test against a supported Mac model running the latest OS X 10.11 security update (2018-001 El Capitan) and let us know the results. I will go ahead and close the issue once we hear back about this part.

Mraccountname, 2018-03-09:
[+] SUCCESS - You are running the latest build number (15G19009) of the OS version you have installed (10.11.6)

I get it now. I did test the build number, and it is fixed.

bruienne, 2018-03-09:
Thanks for confirming, I'll close the issue.

closed 2018-03-09 by bruienne