duo-labs/EFIgy #20: Exception if running on firmware with "E" in build number

sshambar, 2018-01-21:
Running the app on MacBookAir3,2 with firmware "MBA31.0061.B0E" -- received the exception:

```
Traceback (most recent call last):
  File "./efi.py", line 623, in <module>
    efigy_cli()
  File "./efi.py", line 367, in __call__
    self.check_fw_versions(sys_info, api_results)
  File "./efi.py", line 538, in check_fw_versions
    api_efi_build = int(api_efi_str[2].replace("B",""))
ValueError: invalid literal for int() with base 10: '0E'
```
Clearly code isn't handling this build number correctly... should it just be converting from hex?

MyNameIsMeerkat, 2018-03-09:
Thanks for letting us know about this, and yes the build number on Apple EFI's is a base-16 representation and not base-10. 

The fix has been pushed in this commit https://github.com/duo-labs/EFIgy/commit/75705da839eedbed58eea47f91373571eaad4443 let us know if this fixes the problem you are seeing.

Thanks again

MyNameIsMeerkat, 2018-03-21:
OK I'm going to close this out, if the commit did not solve the issue please feel free to reopen

closed 2018-03-21 by MyNameIsMeerkat