duo-labs/EFIgy #21: API returning out of date results

clong, 2018-02-01:
Using the [efigy extension for osquery](https://github.com/trailofbits/osquery-extensions/tree/master/efigy)

```

hostname | build_number_status | efi_version     | efi_version_status | latest_efi_version | latest_os_version | os_version
host-1   | failure             | MBP132.0238.B00 | failure            | MBP132.0233.B00    | 10.13.2           | 10.13.3
host-2   | failure             | MBP142.0169.B00 | failure            | MBP142.0167.B00    | 10.13.2           | 10.13.3
```

The API seems to think 10.13.2 is the most up to date OS version, and the latest_efi_version also seems to be out of date.

bruienne, 2018-03-08:
Hey there! We found the issue that could cause out of date results in certain corner cases and patched the API to fix it. If you could, please test whether the returned results for your endpoints are now as expected.

The dataset is up to date with macOS 10.13.3 Supplemental Update (build 17D102 for non-iMac Pro and 17D2102/17D2104 for iMac Pro endpoints). Thanks for your report!

clong, 2018-03-08:
It appears to be fixed now, thank you!

bruienne, 2018-03-08:
Thanks for reporting back, I will go ahead and close this issue.

closed 2018-03-08 by bruienne