duo-labs/EFIgy #22: Using EFIgy with Mac management systems

jelockwood, 2018-03-21:
It would be nice to have a way to interrogate an entire fleet of Macs via a Mac management system such as JAMF or MunkiReports.

As I see it this would be possible by first installing on _each_ individual client both the python script and the required dependency of pyobjc, one could then via the chosen management/reporting solution run a script which runs the python client on each client.

However it would be preferable if there was a way to do this _without_ having to install anything on the client Macs.

Is there any way to achieve this currently? Perhaps by running a script from the management system and passing the results as variables as part of a URL to https://check.efigy.io the results from the web app would then be scraped and stored in the management system.

bruienne, 2018-03-21:
Hi there! We have a batch mode option in the EFIgy client that would probably work well for this purpose. There is a section in the README that covers this: https://github.com/duo-labs/EFIgy#batch-mode

The idea is to use a management tool like JAMF, MunkiReports, osquery or similar to gather the needed data from each endpoint and then using the EFIgy client to submit a JSON-formatted batch all at once. The API would return results for each endpoint as JSON as well, ready for post-processing. This process could run as a scheduled job that gathers up all the data from newly added endpoints since the last run, at an interval that makes sense for your organization.

Alternatively if you are already using osquery in production you might want to check out [Trail of Bits' osquery EFIgy extension](https://github.com/trailofbits/osquery-extensions/tree/master/efigy) which leverages osquery to run an external query to the EFIgy API on each endpoint through a regular osquery query.

jelockwood, 2018-03-22:
Thanks for your response. osquery with the EFIgy extension does indeed look like what I was looking for. I will still need to install something on the clients but at least I don't need to worry about trying to automate python dependencies for installing. I can then use JAMF to script running osquery.

closed 2018-03-22 by jelockwood

futureimperfect, 2018-03-22:
@jelockwood For what it’s worh, you shouldn’t need to worry about installing any dependencies for EFIgy, as long as you use the system Python interpreter at `/usr/bin/python`, since macOS ships with PyObjC support. Is that not working for you without first installing PyObjC?

jelockwood, 2018-03-22:
I must confess I did not try it first, I was being very literal in following the instructions. However as mentioned the osquery approach looks more promising anyway and I am now looking at its other capabilities including Linux and Zentral.

futureimperfect, 2018-03-22:
Sounds good. Thanks @jelockwood!