duo-labs/EFIgy #23: EFI Verification

dbl001, 2018-04-08:
Is there a way to determine that my EFI  (e.g. MBP53.00AC.B03) has not been comprised?
I tried to reinstall 'MBP53.00AC.B03', however, the reinstaller fails: 

[image]

MacBook Pro (15-inch, Mid 2009) | MacBookPro5,3 | MBP53.00AC.B03 (EFI 1.7) |  

```
$ ./EFIgyLite_cli.py -o

EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Mar 1 2018, 22:25

--------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system information (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : XXX
	Hardware Version : MacBookPro5,3
	EFI Version      : MBP53.00AC.B03
	SMC Version      : 1.48f2
	Board-ID         : Mac-F22587C8
	OS Version       : 10.11.6
	Build Number     : 15G20015

[?] Do you want to continue and submit this request? [Y/N]  y

	Unknown model of Mac supplied: MacBookPro5,3
--------------------------------------------------------------------------------

```

alvarnell, 2018-04-08:
I was going to suggest upgrading to macOS High Sierra, where an EFI verification check is run at every reboot, but I see your MBP can't do that.

It's normal not to be able to reinstall an update that has already been executed.

EFI compromises are extremely rare from what has been reported to date, probably limited to nation-state attacks. What leads you to suspect yours has been.

dbl001, 2018-04-09:
>> What leads you to suspect yours has been.
I’m not sure.  But, ...

In 2015, the network where I was staying was compromised.  
One Linux machine was infected by a root-kit virus.  My Mac appeared unaffected … Virus checkers and malware software didn’t detect anything on my Mac,
However, my LinkedIn account was hacked and highly personal information was added to my profile.
Since, then, there have been a few suspicious OS X crashes (only occasionally).

Could a key-logger have been installed during the 2015 breach.  Could my passwords have been captured?
Some other breaches?  Perhaps an Advanced Persistent Threat (APT)?

ClamAV doesn’t find anything.  Neither did Sophos or MalwareBytes Anti-Malware.

Is there a way to over-ride the variable which prevents me from reinstalling the EFI?
Is there a way to detect whether the EFI has been comprised?
Would ‘netstat' show any unauthorized connections?

> On Apr 8, 2018, at 3:58 PM, Al Varnell <notifications@github.com> wrote:
> 
> What leads you to suspect yours has been.

alvarnell, 2018-04-09:
I'm sure there is a way to hack the installer and prevent the script from checking the current version, but I don't personally know exactly how.

A few years back I was following the UEFI attack details and installed the kext from Darwin Dumper described in this article: [https://www.imore.com/uefi-attack-and-the-mac-what-you-need-know](url). I successfully extracted the EFI but there didn't have anything to compare it to, so didn't pursue the idea. Apple has blacklisted the DirectHW.kext in recent macOS versions, so not sure it will work for you and again you will need something to compare it to.

dbl001, 2018-04-09:
> A few years back I was following the UEFI attack details and installed the kext from Darwin Dumper described in this article: https://www.imore.com/uefi-attack-and-the-mac-what-you-need-know <x-msg://16/url>. I successfully extracted the EFI but there didn't have anything to compare it to, so didn't pursue the idea. Apple has blacklisted the DirectHW.kext in recent macOS versions, so not sure it will work for you and again you will need something to compare it to.

Any comments on using NordVPN?
TOR Browser?
CloudFlare DNS?

https://www.theverge.com/2018/4/1/17185732/cloudflare-dns-service-1-1-1-1

bruienne, 2018-04-09:
Hi there @dbl001 ! Thanks for reaching out with your question. So the nature of EFI/firmware exploitation is that if the attacker does it _right_ you'd never be able to tell during runtime (i.e. what you were looking at) whether anything was modified. An attacker would likely anticipate attempts to verify the firmware's authenticity and return the expected hash when another tool asks for it. One would have to physically extract the firmware from the flash storage (while powered off) in order to do proper consistency checks. This is a complicated process not easily performed by the average or even advanced computer user.

As for your question regarding reinstallation, the EFI updater only allows for incremental version updates which means that it will ignore update requests for the same or older versions of the firmware. Since all firmware payloads are signed it's sadly not possible to "trick" the EFI updater by manually increasing the version number in the payload.

Side note: we only track versions for Macs new enough to be included in the "new" update mechanism that slipstreams them into an OS or Security update. The update you screen captured is the "old" mechanism for the MacBookPro5,3 which is a 2009 Core 2 Duo model that is no longer updated by Apple.

I'm closing this issue as it does not directly pertain to the functionality of EFIgy - feel free to come chat more about this in the #security channel on the Macadmins Slack at [macadmins.org](http://macadmins.org) though!

closed 2018-04-09 by bruienne