duo-labs/EFIgy #25: Request - additional option to auto run

jelockwood, 2018-07-09:
Currently the EFIgyLite_cli.py tool requires interactive response to the question 'Do you want to continue and submit this request? [Y/N]'

This makes it difficult to impossible to 'call' the EFIgyLite_cli.py script via another script.

I therefore request an additional command line option be added e.g. -a to automate or -y to always say yes. Therefore calling the script as follows -

EFIgyLite_cli.py -o -y

would run it and not ask the normal submit question and would return the desired results.

jelockwood, 2018-07-09:
Turned out to be simpler than I thought to add this myself. (I am not a programmer, my skills are mainly shell script but I managed to deduce how to do this in python from the existing command line options so as to add another one.)

Below is a diff between the current standard version and a version with a -a option added to automate the script.

[diff.txt](https://github.com/duo-labs/EFIgy/files/2175872/diff.txt)

Let me know if you need any clarifications.

futureimperfect, 2019-06-14:
Thanks for the feedback, @jelockwood. We've added `-y` as an option in 3c5e59d0b8f46ee9e931a535b10f62b071383610.

closed 2019-06-14 by futureimperfect