duo-labs/EFIgy #26: Support the 2018 MacBook Pros

0xmachos, 2018-08-03:
There's currently no data for the 2018 MacBook Pros. 

Running `EFIgyLite_cli.py` produces the following output: 

```
$ ./EFIgyLite_cli.py 

	Hashed SysUUID   : ****
	Hardware Version : MacBookPro15,2
	EFI Version      : 15.6703.0
	SMC Version      : None
	Board-ID         : Mac-827FB448E656EC26
	OS Version       : 10.13.6
	Build Number     : 17G2208

[?] Do you want to continue and submit this request? [Y/N]  y

	# Results:
		[!] ERROR - Unknown model of Apple Mac 'MacBookPro15,2' given. Provide models in the form of 'MacPro6,1' for Mac Pro (Late 2013) (1533323738.95)

	Highest build number check:
		[-] ATTENTION - You are NOT running the latest release build number of your OS version (10.13.6). Your build number is 17G2208, the latest release build number is 17G65

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.13.6)
```

While I'm here, thank you for this research the API and paper are fantastic.

0xmachos, 2018-08-03:
I think on the 2018 MBPs the same issue will exist as on the iMac Pros with reading the SMC version. This is mentioned on line [416-419](https://github.com/duo-labs/EFIgy/blob/master/EFIgyLite_cli.py#L416-L419) `EFIgyLite_cli.py`. 

This is purely based on that comment and the above output reporting:  
```
SMC Version      : None
```

jelockwood, 2018-08-06:
The SMC issue as documented regarding the iMac Pro does indeed also apply to the new MacBookPro15,1 however I am not convinced that should make it impossible to return a valid result - when EFIgy is updated to recognise the new MacBookPro15,1.

As the name of the program says, it is the EFI i.e. firmware version we are really interested in and not the SMC version. It maybe the SMC version helps the authors in their checking.

It should also be noted that the EFI version returned by EFIgy on the MacBookPro15,1 is not the same format as System Information shows or system_profiler returns.

Have a look at the result of -

system_profiler SPHardwareDataType | grep "Boot ROM"

This gives a result of "Boot ROM Version: 15.16.6703.0.0,0" whereas EFIgy lists this as "EFI Version      : 15.6703.0" and is chopping of the later part of the result.

cemb007, 2018-10-20:
While technically not the same model, would it be possible to update and include checking for MacBookPro Retina 2012?  I get the following when run with the debug flag -

MacBookPro:EFIgy $ ./EFIgyLite_cli.py --debug
DEBUG:__main__:[-] Certifi module not found, falling back to bundled cecert.pem file
DEBUG:__main__:[+] cacert file location: '/Users/cembasu/Downloads/EFIgy/cacert.pem'

EFIgyLite API information:
	Server: https://api.efigy.io
	API Version: 0.2
	Updated On: Mar 1 2018, 22:25

------------------------------------------------------------------------------------------------------------------------------------
Endpoint: 127.0.0.1
	# Enumerated system information (This data will be sent to the API in order to determine your correct EFI version):

	Hashed SysUUID   : fac95e86611f96a00c046b20c60148fbb919f555d1c738ec3a6ef47d5544b0e1
	Hardware Version : MacBookPro10,1
	EFI Version      : 251.0.0
	SMC Version      : 2.3f36
	Board-ID         : Mac-C3EC7CD22292981F
	OS Version       : 10.14.1
	Build Number     : 18B67a

[?] Do you want to continue and submit this request? [Y/N]  y

	# Results:

	EFI firmware version check:
		[!] ERROR - Unknown Build Number '18B67a' given. (1539995499.06)

	Highest build number check:
		[!] ERROR - Unable to find build number for supplied os version 10.14 (1539995499.06)

	Up-to-date OS check:
		[!] ERROR - Unknown OS version '10.14' given. Provide OS versions in the form of <Major>.<Minor> e.g. '10.12'. (1539995499.06)
------------------------------------------------------------------------------------------------------------------------------------

Thank you for providing this insightful check.

alvarnell, 2018-10-20:
You are running a beta version of macOS, so you won't be able to check that MBP for valid EFI until the release version of 10.14.1 comes out and the folks here update their EFI database.

-Al-

futureimperfect, 2019-06-14:
Hi @0xmachos,

Sorry for the delay. This should be resolved now, but feel free to reopen if not.

Thanks!

closed 2019-06-14 by futureimperfect