duo-labs/EFIgy #28: Handling for new firmware version style seems to be incorrect

swrobel, 2019-06-14:
Using both check.efigy.io and EFIgy GUI I get this mismatch:

Your EFI version is not what we expected.
You're running 166.0.0.0.0, but we expected you to be running IM171.0166.B00.

The CLI does not report the expected version, but says `Your firmware version 166.0.0 is older than expected for a iMac17,1 running build 18F132. You should update your EFI firmware`

Model: iMac17,1
Build: 18F132

alvarnell, 2019-06-14:
Apple changed the way it displays the EFI version, so this may be an issue for everybody now.
In @swrobel's case, the 166 part matches 0166, and so should be told he is up-to-date.
This article explains the change which occurred last October: [EFI firmware changes with 10.14.1, HS Security Update 2018-002, and Sierra Security Update 2018-005](https://eclecticlight.co/2018/10/31/efi-firmware-changes-with-10-14-1-hs-security-update-2018-002-and-sierra-security-update-2018-005/)

swrobel, 2019-06-14:
There does seem to be [code added ](https://github.com/duo-labs/EFIgy/commit/3c5e59d0b8f46ee9e931a535b10f62b071383610#diff-eebe91fe7320688945745dcc1feb0fb6R571)to handle this, but it doesn't seem to be working correctly ... at least in this case.

futureimperfect, 2019-06-16:
Hi @swrobel and @alvarnell,

Are you seeing these issues with the latest version of the EFIgy client? If I change add the following at line 396 I'm still seeing `SUCCESS` all around.

```
sys_info['hw_ver'] = 'iMac17,1'
sys_info['rom_ver'] = '166.0.0'
sys_info['build_ver'] = '18F132'
```

Also, you're right that check.efigy.io and EFIgy-GUI haven't been updated to parse the newer version format, but I just added that to EFIgy-GUI here:

https://github.com/duo-labs/EFIgy-GUI/commit/c1428a8c5e262077abbb0559da62f6328bf08ea8

I haven't released a new version yet but if you feel so inclined go ahead and try the updated build. I'll try to release an updated version soon.

Thanks!

swrobel, 2019-06-16:
@futureimperfect yes, I tested with latest master & just checked the code to verify and it's still failing with `Your firmware version 166.0.0 is older than expected for a iMac17,1 running build 18F132. You should update your EFI firmware`. Is there some way to show the build number that EFIgyLite expects?

alvarnell, 2019-06-17:
@swrobel, the master hasn't been updated yet. You will need to substitute https://github.com/duo-labs/EFIgy/blob/master/EFIgyLite_cli.py in order to use the new changes.

futureimperfect, 2019-06-17:
Hi @swrobel and @alvarnell,

You're totally right. The updates I made recently didn't take into account the newer EFI version style when checking EFI only. I just updated the dataset again, so it should work now. Feel free to reopen this issue if not!

Thanks!

closed 2019-06-17 by futureimperfect