duo-labs/EFIgy #29: Version information in EFIgyLite_cli.py wrong

jelockwood, 2019-06-17:
Having over the weekend received messages indicating a recent update to the script to make it compatible with Apple's new firmware versioning approach I have just visited the github to look at the new script.

A very quick initial test does suggest it is working better.

I did however notice that the date and version information in the script is totally unchanged which along with the fact the first page or so of code is unchanged would lead a less observant person to think the entire script is unchanged. (It is changed. :) )

Can I request you tidy it up by updating this information.

An optional consideration would be that I had added a custom change to the last version to make it more usable for my own purposes but it would possibly help others. I added a -a command flag to get it to run automatically with no user interaction. This allowed me to call it from a management tool to get a report.

alvarnell, 2019-06-17:
The master has not yet been updated with the new script, which should explain why the date and version information hasn't been updated yet.

I would think that the new -y option serves the exact same purpose as your suggestion for -a.

jelockwood, 2019-06-17:
@alvarnell 
Thanks, yes the -y seems to do what I want 👍

closed 2019-06-17 by jelockwood

futureimperfect, 2019-06-18:
> The master has not yet been updated with the new script, which should explain why the date and version information hasn't been updated yet.

Hi @alvarnell,

I'm not sure what you mean by this. The `master` branch is definitely up-to-date and we haven't released EFIgy anywhere else besides GitHub. Can you elaborate on what you mean?

Thanks!

alvarnell, 2019-06-18:
Sorry, when I downloaded the master yesterday (or perhaps the day before) it was identical to the one I already had from over a year ago, but I see today that it has been updated, so the date and version number comment is relevant.