duo-labs/EFIgy #30: -o command line flag returns bad result

jelockwood, 2019-06-17:
With the latest version of the script if I do -

EFIgyLite_cli.py -o

Which is only supposed to return the firmware information I get the following result

	Your firmware version 194.0.0 is older than expected for a MacBookPro14,3 running build 18F132. You should update your EFI firmware.

If I do not use the -o option then I get

	# Results:

	EFI firmware version check:
		[+] SUCCESS - The EFI Firmware you are running (194.0.0) is the expected version for the OS build you have installed (18F132) on your MacBookPro14,3

	Highest build number check:
		[+] SUCCESS - You are running the latest build number (18F132) of the OS version you have installed (10.14.5)

	Up-to-date OS check:
		[+] SUCCESS - You are running the latest major/minor/micro version of the OS you have installed (10.14.5)

This discrepancy might explain why @swrobel is seeing a problem in issue https://github.com/duo-labs/EFIgy/issues/28

alvarnell, 2019-06-17:
Confirmed here.

futureimperfect, 2019-06-17:
Hi @jelockwood  and @alvarnell,

The updates I made recently didn't take into account the newer EFI version style when checking EFI only. I just updated the dataset again, so it should work now. Feel free to reopen this issue if not!

Thanks, and good catch!

closed 2019-06-17 by futureimperfect