duo-labs/EFIgy #33: https://api.efigy.io/ SSL Certificate has expired

MLBZ521, 2020-10-21:
The SSL Certificate for https://api.efigy.io expired on September 17, 2020.

Not sure if Duo is still even supporting this utility, doesn't seem to have updated any models/firmware since Mojave.

Will likely have to retire the use of using this utility in my environment for checking the firmware state.

alvarnell, 2020-10-21:
The last time I checked the lead developer had moved on to Apple and I believe the only other person that worked on the project also left, so I would certainly consider this to be abandonware.

jelockwood, 2020-10-21:
The source of the client is (obviously) available here. Does that give enough information for someone to create a replacement for the backend i.e. api.efigy.io?

Without having looked at the entire client I would be guessing the backend is merely a database of values.

This then leaves just the matter of adding new data which admittedly due to Apple's silence is not a small task.

MLBZ521, 2020-10-22:
Yeah, I was noticing the the contributors appeared to have moved on.

It may be possible @jelockwood.  I looked through everything a while ago when I wrote a process that utilized this utilities functionality, but that was some time ago and I'd have to look into it.  Outside of that, someone would have to be willing to host and maintain that data and be willing to service and accept the incoming connections.  Regarding the data itself, I would consider this as "crowd sourced" verified; the likelihood of ever seeing something again from Apple regarding official firmware version/per OS build/per HW model is very unlikely..  That said, last I checked, that information can be extracted from the OS updates themselves, but again, that's not a small task either, but maybe something that could be done...  I'll add it to my rainy day project list to look into.  Though, discussing with my colleague at work, this really probably isn't needed any more as Apple is (albeit Apple as the source) claiming they're managing firmware better/more strictly.

alvarnell, 2020-10-22:
FYI, there are already a couple of sources that might be useful as currently maintained databases.

All current Software Update catalogs have links to the current EFIAllowListAll.pkg which contains the EFIAllowListAll.bundle with listings of every EFI update ever released for all Mac models. It's used by the `eficheck` process conducts a weekly check and notifies the user if there's an issue with either an unknown version or changed hash value. Generally speaking, the highest version listed is the latest version.

Howard Oakley maintains this database which lists the latest **release** version for each Mac model [](https://eclecticlight.co/2019/10/08/which-efi-firmware-should-your-mac-be-using-version-4/). Beta test users may have a more recent version, which was always an issue for this tool, as well.

jordan-wright, 2020-10-22:
Hey there!

Thanks for sending this in. I'm checking internally to determine how we want to proceed with the EFIgy service. You're correct that it isn't as maintained as we would like, so I'm going to figure out what a path forward looks like so that we can set expectations for folks using the service.

I'll keep y'all updated!

MLBZ521, 2020-10-22:
@alvarnell Yeah, I think that .pkg is what I was thinking of/referring to in my last post.  Thanks for the extra info as well.  👍. Depending on what Duo does, I may look more into this and how to handle if we want to continue to monitor the firmware.

@jordan-wright Thank you sir! 👍

n8felton, 2021-03-26:
Just adding a note that it looks like @jordan-wright moved on from Duo as well now.